patience4711/RPI-APS-inverters #112: using mosquitto in home assistant

defaliz, 2024-04-11:
Hi,
i'm using mqtt to create sensors in Home Assistant
i have datas in mqtt like this : 
{"serial":"xxxxxxxx","acv":232.6,"freq":50.0,"temp":15.1,"pwr":[21.4,20.6],"en":[41.6,40.6],"totalpwr":42.0,"totalen":82.2}
with differents lines and serial, because i've 3 inverters
but how can i discriminate datas between my 3 inverters ?
thank a lot
here is a declaration sensor in my home assistant : 
mqtt:
  sensor:
#power  
    - name: inv0 pwr1
      unique_id: inv0 pwr1
      state_topic: "rpiecu/out"
      json_attributes_topic: "rpiecu/out"
      value_template: '{{ value_json.pwr[0] }}'
      unit_of_measurement: 'W'
      
    - name: inv0 pwr2
      unique_id: inv0 pwr2
      state_topic: "rpiecu/out"
      json_attributes_topic: "rpiecu/out"
      value_template: '{{ value_json.pwr[1] }}'
      unit_of_measurement: 'W'

defaliz, 2024-04-12:
here is result in mqtt explorer, 3 lines (first none panel is connected)
11:45:27
{"serial":"703000518414","acv":0,"freq":0,"temp":0,"pwr":[0,0],"en":[,],"totalpwr":0,"totalen":0}
11:45:24
{"serial":"703000557318","acv":238.7,"freq":50.0,"temp":28.6,"pwr":[255.2,246.5],"en":[416.4,392.7],"totalpwr":501.7,"totalen":809.0}
11:45:20
{"serial":"703000071646","acv":238.7,"freq":50.0,"temp":31.3,"pwr":[189.7,331.8],"en":[340.0,565.4],"totalpwr":521.5,"totalen":905.4}
is there a way to create 3 differents sensors in Home Assistant... ?
regards

patience4711, 2024-04-12:
I am not really familiair with Home Assistent but i think that each sensor has its own mqtt topic. <br>
You can have different topics for each inverter in the RPI-ECU. First you go to the inverter settings and fillup an IDX number for each inverter (p.e. 111, 222, 333 if you have 3 inverters). <br>Now, in mqtt settings, put a slash ("/") behind the send topic .<br>Now the idx of the inverter will be added automatically to the sendtopic. P.e. if you have an inverter with IDX **123**, and a sendtopic **your/topic/** the mosquito message goes to **your/topic/123**.
This way you can have a separate topic for each inverter.

defaliz, 2024-04-12:
thanks again !
it's exactly what i want and that works like a charm...
regards
Alain

closed 2024-05-08 by patience4711

autourdupc, 2025-01-05:
@defaliz 

Hello defaliz.
I just start with HA, and try to link RPI-ECU with HA via MQTT.

I really need help to get the data... Could you explain to me ?
I just installed MQTT in HA, then never be able to get data from RPI-ECU.
[image]

Thanks in advance.

defaliz, 2025-01-06:
Hi autourdupc
do you create IDX for each inverters and mosquitto host and port in rpiecu interface ?
after that you must create entries in a yaml file, and you should see sensors in HA
sorry but, i'm far from my home, and for now there is a big storm in Normandy and problem on enedis power...
so no remote access... as soon as power is establish, i'll send you my scripts
regards

autourdupc, 2025-01-06:
Hello.

I created IDX for each inverters
Concerning mosquitto, here are the settings
[image]

defaliz, 2025-01-06:
i think you must not add http for address mqtt, because this is not the protocol
simply the ip address
here is my conf, in ecu_data/mqttConfig.json file

{"enabled":1,"retain":0,"address":"10.3.141.16","port":1883,"intopic":"rpiecu/in/","outtopic":"rpiecu/out/","auth":1,"username":"mqtt","password":"mqtt","format":2}

autourdupc, 2025-01-13:
@defaliz 
I love creating programs... But here, HA is very strange !
Even trying to understand how it works, I'm unable to get my sensor running without error... I find HA very very hard to use !!!

autourdupc, 2025-01-13:
Thanks for the help...
I was able to get tests messages from RPI-ECU.
Today, no messages anymore, nor new sensor appeared despite my settings... Very boring !

defaliz, 2025-01-13:
hum...
connecting with ssh on rpiecu can you publish and subscribe on HA with 
mosquitto_pub and mosquitto_pub ? 
mosquitto_sub -v -h ip_HA  -u username -P password -t '#'

autourdupc, 2025-01-13:
Using test button I get the output on a new page...
running script mqttTest...
the value of auth = 1
the value of username = MQTT_User
the value of password = mqtt
testcommand = mosquitto_pub -h 192.168.20.241 -u MQTT_User -P mqtt -t rpiecu/out/ -m 'authenticated hello from rpi ecu'
mosquitto test runned..
HTTP:1/1 200 OK

defaliz, 2025-01-13:
i think you found a bug...
further to test, i've used mqtt explorer to see messages...
and there are only my last test from rpiecu, and no history from inverters
BUT, my sensors rpiecu on HA are all available ! i don't understand...
last week i used version 3_10 and i don't verify...
maybe a problem with version 4...
we must wait for tomorow having new solar production...
[image]

defaliz, 2025-01-13:
faudra ouvrir un case demain si ça persiste...

defaliz, 2025-01-14:
hello,
strange i don't know why but for now on mine i can only seeing mqtt test
[image]
you are on RPI-ECU-v4_3 ?
can you please send me contain of /var/www/ecu_data/mqttConfig.json
merci

autourdupc, 2025-01-14:
I can get values from RPIECU, only in MQTT Explorer
Nothing here...
[image]

Here is my /var/www/ecu_data/mqttConfig.json file content
{"enabled":1,"retain":0,"address":"192.168.20.241","port":1883,"intopic":"rpiecu/in","outtopic":"rpiecu/out","auth":1,"username":"MQTT_User","password":"mqtt","format":2}

defaliz, 2025-01-14:
thank
in HA i don't use configuration of mqtt
[image]
i only create sensor in mqtt.yaml, restart HA and tha't all

defaliz, 2025-01-14:
so, i've launch manually
/usr/lib/cgi-bin/ecu/autoPoll.pl

and now i've good message from rpiecu
[image]
and here is datas in HA
[image]

do you create your sensors in configuration.yaml or in a separate file ?

autourdupc, 2025-01-14:
I think you are right...
There is no auto refresh after you lauch a "Test" un RPIECU
[image]

autourdupc, 2025-01-14:
> do you create your sensors in configuration.yaml or in a separate file ?

In a separate file... But I think I made a mistake in the folder tree... I need to check

defaliz, 2025-01-14:
first you can try to put your rpiecu.yaml in same folder than configuration.yaml
sensors w'll be seing in mqtt integration
[image]
have a look to your logs in système, journaux
[image]

autourdupc, 2025-01-14:
The MQTT Broker logs...
[image]

defaliz, 2025-01-14:
You have same logs as me for mqtt, so maybe problem in sensors creation...
you have to restart HA each time you manually modify a yaml file, if there is a format problem in the yaml file HA you'll know it...
i've read autoPoll.pl (extract in bottom of message ) and there is no poll of mqtt during nighttime and if polling of inverter failed (i understand why i hav'nt no message yesterday :) )
I post you a message on your issue "https://github.com/patience4711/RPI-APS-inverters/issues/129" because there is something strange...

 __my$autopollCmd = "sudo /usr/lib/cgi-bin/ecu/inverterPoll.cgi $_";
          print "autopollCmd = $autopollCmd\n<br>";
          my $result = system($autopollCmd);
          print "result = $result \n<br>"; # result = 0 or 256 -1

          # if result = 0 the polling succeeded then we should mqtt
             if($result == 0)
             {
                 my $mqttCommand = "/usr/lib/cgi-bin/ecu/sendMqtt.cgi $_";
                 system($mqttCommand);
             }__

autourdupc, 2025-01-14:
I did not check on existing files in RPIECU... I do not have time to make my HA running, do my job (hey, need also to earn money !), debug (if needed RPIECU)... 
But I think I got it after pressing "Test" on RPIECU... It seems to stop / fail polling sometimes !

defaliz, 2025-01-14:
Yes mine failed checking zigbee device very often, and then no polling at all ..
I've used rpiecu.3_4 version for 2 years without this problems...