patience4711/RPI-APS-inverters #116: Chart for

defaliz, 2024-05-30:
Hi
i'm wondering why is there no charts for totals values ( power and energy) like for inverters ?
maybe on a next version  :-)
regards
[image]

patience4711, 2024-06-12:
That graph will look the same as all individual graphs so it yields nothing new. It would take too much time to query all the inverter databases and totalize all the values. So no, there are no plans to implement that.
Maybe your domoticz software can do it.

closed 2024-06-28 by patience4711

autourdupc, 2024-11-18:
Hello @defaliz...

Maybe you could try this one I've just created... 
https://github.com/patience4711/RPI-APS-inverters/issues/129