patience4711/RPI-APS-inverters #117: No back button in the Memo page (V3.10)

autourdupc, 2024-07-12:
Hello patience4711.

Nice to contact you again for this fabulous project.
Due to SD card corruption I loose all my job and datas (and no backup) !

Just start again with V3.10.
When in Memo page, there is no button to return back to main menu.

[image]

Regards,
Laurent.

patience4711, 2024-07-16:
Hi Laurent, It's a shame to lose the data this way. It also happened to me some time ago, it is really important to have a good quality sdcard. Luckily, you didn't lose the kWhr's. 
About all the troubles still remaining:  I am running  myself  a brand new version with the latest linux version (bookworm). So far i hesitated to publish that, it means that one has to start all over with burning a new image on an sd card.  
As far as I can see now there are still a few teething problems so if I can find time I make it allright and i'l publish an image of the new system. The zigbeecommand however works correctly. I hope within a few weeks i can come up with the new version.
So please be patient..
Best regards
Hans

autourdupc, 2024-07-23:
Hi Hans.

I'm happy to know that you are working on a new version... I still hope I can provide a new main page with more stuff... But I lack time for it, and need to do some reverse engineering in your code to understand some stuff and learn how to deal with the database. 
Your job is so great so I need to be very good for my project.

FYI, I got difficulties to pair inverters, and found that they need 40 w or more per panel to correctly pair. (QS1 with 20 w per panel do not pair correctly, till the wattage increase to 40 w). 

Concerning new version, I will be patient...

patience4711, 2024-07-23:
Today i made it available.
It is know that pairing works better with a little sunshine, hard to explain why. Interesting thay you found out that there seems to be some treshold value for the pairing. I can mention that in the wiki, thanks.