patience4711/RPI-APS-inverters #118: Resetting database do not reset all values

autourdupc, 2024-07-13:
Hello patince4711.

After a newly installation of V3.10, and of course a database reset, after few hours of collecting data during a poor weather conditions, I got remaining values not depending from my solar installation (overall total energy value).

[image]

patience4711, 2024-07-19:
This would mean that the database was not cleared after the factory reset, Those values are calculated with q=values from the existing databases. You can check in the database console if there are existing values in the database from before your installation. Or  It that's the case you can try to wipe them. In the future new version i'l test the reset function thoroughly. 
Or maybe the page you are viewing is cached by your browser, you can try to refresh the page.

autourdupc, 2024-07-23:
Its not web browser cache because got it on many devices...
I think there a re values remaining in txt files you use for calculations... So database reset needs also txt files reset.

patience4711, 2024-07-23:
No textfiles used for saving data like that, the values are computed with database queries  every time you open that page. Only the values of currently existing inverters (that have a file var/www/ecu_data/inverters/invPropertiesx) are used. 
If the database is empty the values are all null.

You say you have this problem on many devices? Could you explain that a bit more?

If you did the factory reset, all databases and inverter entries are wiped so there can't be any remaining values somewhere i guess.
With the newest version 4_0 i added some inverters and imported a database from my working system. After factory reset this was all gone.  So for me it works well, that makes it impossible to say what could be wrong in your case. 
[image]

patience4711, 2024-07-30:
As to playing you should know that there is an important revision in the way the scripts work. Since there is no wiringPi anymore in the new linux version, i had to revise all the scripts that use the zigbee module (com port). 

A good moment to streamline this. So i made one script in which all io related functions are concentrated. This script is included by all scripts that need to communicate with the zigbee module. (libraries/serial_IO.h)

Happy playing..

closed 2024-11-11 by patience4711