patience4711/RPI-APS-inverters #119: no polling since 2 days

defaliz, 2024-07-14:
Hi,
since 2 days there is no polling on my raspberry... version ECU 3.10
when i poll inverters one by one with the web interface it's OK,
but when i launch polling with the command on CLI, like in crontab...
/usr/lib/cgi-bin/ecu/autoPoll.pl

i've this error
Can't locate DateTime/Locale.pm in @INC (you may need to install the DateTime::Locale module) (@INC contains: /etc/perl /usr/local/lib/arm-linux-gnueabihf/perl/5.28.1 /usr/local/share/perl/5.28.1 /usr/lib/arm-linux-gnueabihf/perl5/5.28 /usr/share/perl5 /usr/lib/arm-linux-gnueabihf/perl/5.28 /usr/share/perl/5.28 /usr/local/lib/site_perl /usr/lib/arm-linux-gnueabihf/perl-base) at /usr/lib/arm-linux-gnueabihf/perl5/5.28/DateTime.pm line 16.
BEGIN failed--compilation aborted at /usr/lib/arm-linux-gnueabihf/perl5/5.28/DateTime.pm line 16.
Compilation failed in require at /usr/lib/cgi-bin/ecu/autoPoll.pl line 4.

nothing new has been installed on this raspberry...
regards

defaliz, 2024-07-15:
i've comment use date time in 
/usr/lib/cgi-bin/ecu/autoPoll.pl
and know that works, but i don't think it's a good thing...

#!/usr/bin/perl
use strict;
use warnings;
#use DateTime;

patience4711, 2024-07-20:
If you didn't touch the system and this happened 'out of the blue', i wouldn't know what could be wrong. 

This seems not possible to me. 

Anyway, if it works oke now than you don't have to worry about that date/time module.

defaliz, 2024-07-20:
thank for reply...
yes strange things...
i've seen that only autopoll.pl had "use DateTime", and the script doesn't use it.. so no problem...
regards
Alain

closed 2024-07-24 by patience4711