patience4711/RPI-APS-inverters #120: Send a Zigbee command 2710AABBCC fail in Raspberry Pi zer0 2 (V3_10)

autourdupc, 2024-07-15:
Hello Patience.

Seem there is a regression in the zigbee command code...

`running script sendZigbee.pl...
argument is 2710AABBCC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html
running testZigbee.cgi version 2024-jan-12
testZigbee running with arg 2710AABBCC
wiringPi started
slen = 03
command incl. FE, len and crc = FE032710AABBCCD0
no answer
command processed
HTTP:1/1 200 OK`

We got this previously in 3.8 as I remember...

Regards,
Laurent.

patience4711, 2024-07-17:
It is a bit confusing but installing the update should solve this: https://github.com/patience4711/RPI-APS-inverters/issues/103

autourdupc, 2024-07-24:
I downloaded the source code you provided few months ago and compiled it and got now a working command.

closed 2024-07-24 by patience4711