patience4711/RPI-APS-inverters #121: new version 4_0

patience4711, 2024-07-24:
The kernel on the version 3_x is several years old, Sadly it is not possible to issue software updates that also upgrades the linux verson. So i rebuild the system on a new linux release (bookworm). This involved a lot of work and testing because a lot of things work differently in this new version, For example all scripts that use the communication port had to be revised.  

From now on the old versions are not updated anymore because i have no working system with 3_9 so i can't test anything. This doesn't mean that your current system is useless now, when it works you can enjoy it for years.

However if you want to be up-to-date you can start with the new system. The only way is to burn the image to a (new) sd-card, Remember that it is possible to backup and restore the database of your old system.

I have been using the new system for several months without any problems so i consider it well tested.

autourdupc, 2024-08-22:
Hello Patience4711.

I will soon move to new version, and I thought about a new feature...

Are you aware of a backup function like the one we can find in Freepbx distro named "raspbx-backup"  : it backups the complete SD-CARD to an image file.
It works this way : 
1) First run it creates (while system is running in production) an image file of the entire SD-CARD (you pass settings to tell path and name - Can be elsewhere on a mapped drive or FTP...)
2) Each new run (while system is still running in production) updates the existing image file (I personnaly use a cron job to update remotely on an FTP outside server my Freepbx SD-CARD)
  -> The image file is exactly the mirrot of the SD-CARD running, so its easy to duplicate / create a new SD-CARD in case of corruption without any settings.

I think there is an open project doing this raspbx-backup function... It would be cool for all your projets.

Regards,
Laurent.

patience4711, 2024-08-22:
Hi Laurent, this sounds good. There are however some concerns. 
First of all, the image should be saved to an external storage, otherwise it has no use. This means extra hardware and space. And when we have this external storage, it would be better to run the whole system from this medium.  But when this gets corrupted we w'd lose our image too. 

So basically it could work like this:
 - fit a usb stick (need an adapter cable microusb to usb or something)
 - install Asterisk for raspberry pi (no version for bookworm availlable and pi-zero not supported)
-  have a script that runs the backup program every night.

We also need to have a setting that the admin can use to enable/disable the backup. 

Saving the image via ftp is something for only advanced users. 

So this all seems not simply to implement at this moment.

autourdupc, 2024-12-06:
Hello @patience4711 

I've just installed last 4.2 version bec. got an SD card corruption, then restored a previous 4.1 backup (and at this time I did an expand as per requested).

I notice now that boot process is very very long... It takes a while to get the ssh login, same for web index page.
It was not so long in 4.1 version (maybe 4.2a version), I do not really remember.

I found alos that CPU load is highter now 1.5 to 2.5 in Top command, while not poolling at night.

Did you change something that could increase system load ?

Regards,
Laurent.

patience4711, 2024-12-30:
No, the changelog explains what i changed, nothing that could increase the cpu load.

autourdupc, 2024-12-31:
Hello @patience4711 

I think I found some explanations...
Testing my Samsung Evo SD-Card, I found a R/W speed up to 4,5 MB using dd command... This is so poor !

So according to this page
https://bret.dk/best-raspberry-pi-micro-sd-cards/#Raspberry-Pi-Zero-2-microSD-Card-Performance
I ordered a kioxia SD-CARD and will see if it works better when I got it !

closed 2025-02-11 by patience4711