patience4711/RPI-APS-inverters #122: Version 4_0 password for RPI-ECU accespoint

Watt600, 2024-08-03:
First of all, many thanks for the great software. Version 3_9 is running very well and stable for me.

Nevertheless, I wanted to switch to the new version 4_0 now.
I have written the image to an SD card and the RPI is also booting. After a while I see the accesspoint "RPI-ECU". However, the password 12345678 is not accepted!
I have this behavior both from my notebook and from my smartphone. What is the correct password?

patience4711, 2024-08-05:
Sorry to hear that. Below is part of the script that sets-up the accesspoint. Clearly the psk is 12345678 so i really would't know why it does't work. I did test this thoroughy and had no troubles with it. 
Your device says that the password is incorrect? Could you try with an android device? Do you use ipv6?

I am now testing with a setup in wich i skipped some of the security lines. If i have results i'l publish a new image a.s.a.p.

[image]

Watt600, 2024-08-05:
Thanks for the quick reply!

Yes, the password 12345678 is correct.
I have mounted the image and here is the content of the file
rootfs/etc/NetworkManager/system-connections/TEST-AP.nmconnection:

[connection]
id=TEST-AP
uuid=e6643536-cbb9-4242-b964-1fb8fab6320a
type=wifi
autoconnect=false
interface-name=wlan0

[wifi]
band=bg
channel=3
cloned-mac-address=00:12:34:56:78:9A
mode=ap
ssid=RPI-ECU

[wifi-security]
group=ccmp;
key-mgmt=wpa-psk
pairwise=ccmp;
proto=rsn;
psk=12345678

[ipv4]
address1=192.168.4.1/24
method=shared

[ipv6]
addr-gen-mode=default
method=disabled

[proxy]

Here is the output of "dmesg" when I connect to the RPI-ECU.
The line is striking:
wlp3s0: deauthenticating from 00:12:34:56:78:9a by local choice (Reason: 17=IE_DIFFERENT)
After that, my notebook (Linux) successfully reconnects to the Fritz.Box

dmesg:
======
[53347.365079] wlp3s0: deauthenticating from 00:12:34:56:78:9a by local choice (Reason: 17=IE_DIFFERENT)
[53363.086953] wlp3s0: authenticate with 00:12:34:56:78:9a
[53363.120768] wlp3s0: send auth to 00:12:34:56:78:9a (try 1/3)
[53363.122513] wlp3s0: authenticated
[53363.129672] wlp3s0: associate with 00:12:34:56:78:9a (try 1/3)
[53363.141615] wlp3s0: RX AssocResp from 00:12:34:56:78:9a (capab=0x411 status=0 aid=1)
[53363.144081] wlp3s0: associated
[53363.144211] ath: EEPROM regdomain: 0x8348
[53363.144213] ath: EEPROM indicates we should expect a country code
[53363.144214] ath: doing EEPROM country->regdmn map search
[53363.144216] ath: country maps to regdmn code: 0x3a
[53363.144217] ath: Country alpha2 being used: US
[53363.144218] ath: Regpair used: 0x3a
[53363.144219] ath: regdomain 0x8348 dynamically updated by country element
[53363.159329] wlp3s0: deauthenticating from 00:12:34:56:78:9a by local choice (Reason: 17=IE_DIFFERENT)
[53371.245948] wlp3s0: authenticate with c8:0e:14:fritz:box
[53371.292926] wlp3s0: send auth to c8:0e:14:fritz:box (try 1/3)
[53371.295256] wlp3s0: authenticated
[53371.297524] wlp3s0: associate with c8:0e:14:fritz:box (try 1/3)
[53371.300778] wlp3s0: RX AssocResp from c8:0e:14:fritz:box (capab=0x1511 status=0 aid=1)
[53371.302978] wlp3s0: associated
[53371.303101] ath: EEPROM regdomain: 0x8114
[53371.303103] ath: EEPROM indicates we should expect a country code
[53371.303104] ath: doing EEPROM country->regdmn map search
[53371.303105] ath: country maps to regdmn code: 0x37
[53371.303106] ath: Country alpha2 being used: DE
[53371.303107] ath: Regpair used: 0x37
[53371.303108] ath: regdomain 0x8114 dynamically updated by country element
[53371.394087] wlp3s0: Limiting TX power to 20 (23 - 3) dBm as advertised by c8:0e:14:fritz:box

The connection does not work from the Android smartphone either.

Thank you for your effort, I will wait patiently. If any information is still missing, just let me know.

Watt600, 2024-08-05:
Sorry, closed by mistake ;-)

Watt600, 2024-08-05:
That sounds good.

But the mounted image is unfortunately "read only", even if I mount it as root.
There seems to be no way to mount it writeable :(

Thank you very much for the great support!

Watt600, 2024-08-05:
Thanks for the tips on mounting!
The download is running, I'll get back to you.

Watt600, 2024-08-05:
It works!
Now I can port the system to the new version for the coming weekend.

Thanks again for the great support and good luck and good health for the future!

patience4711, 2024-08-05:
Thank you, i wish the same to you. I also made a software upgrade to correct the remaining known issues.

I am still puzzled what the trouble was. Since it works now one should say that the password was the problem. Googling did not help to find out why the password could be invalid, must be the length but oke, if it works, it works.

Watt600, 2024-08-09:
@patience4711 I have installed the new image 4_1. Everything worked great.
The Zigbee pairing with the YC600 was also successful.
The new version is now in operation.

Thanks again for your great work!

patience4711, 2024-08-09:
Good to hear that. Did you also migrate your database?

Watt600, 2024-08-09:
Sorry, I forgot to mention this, but yes, I have also successfully restored the database from the system under v3_10!

Great software! :)

closed 2024-08-10 by patience4711