patience4711/RPI-APS-inverters #123: version 3_10 no chart for chart_pe.php page

defaliz, 2024-08-08:
Hi
thanks for new version 4
i'll try soon...
but for now i've a problem with version 3.10, i've no chart on chart_pe.php
i've seen that the script read /var/www/ecu_data/chartDate.txt but this file does not exist !
how  and when is it create ?
thanks for help
alain

patience4711, 2024-08-09:
It is mainly a linux update. No really new features but the performance of the webinterfaces on telephones is better.

defaliz, 2024-08-09:
thanks a lot...

closed 2024-08-10 by patience4711

defaliz, 2024-08-20:
Hi,
sorry to re-open this post... but the problem persist...
yesterday (19/08) i had chart on this page chartjs/chart_pe.php?page=0&inv=0
but nothing for the others days... today and before yesterday...
how can i investigate furthermore ?
regards

defaliz, 2024-08-24:
ok, sorry, the problem was that influxd was stopped... i don't know why...
sometimes it does...
it's OK know
regards