patience4711/RPI-APS-inverters #124: New version 4.1 crash after booting and RPI power off by itself

autourdupc, 2024-08-23:
Hello patience.

Burned today a new SD-CARD (just out-of-the-box).
I was able to set the Wifi according to my access point.

Next boot is very very long compared to V3.x (> 90 sec)
I was able to get the main page, and clic the "Info" menu, then... Nothing !
The RPI shutdown by itself.

I then try to power on again and got the main menu after a long time again. Got again the main screen then RPI shutdown immediately.

Find attached log folder
[log.tar.gz](https://github.com/user-attachments/files/16725826/log.tar.gz)

I inserted again previous SD-CARD running V3.x, it takes < 30 sec to work fine !

Regards,
Laurent

patience4711, 2024-08-23:
I installed that image often enough to trust that it works ok. At least one other user installed it without problems. So the software seems ok.

What raspberry do you use? How large is your SD-card?

The only thing i can imagine is that you used a too small sd-card (or a bad one).

autourdupc, 2024-08-24:
Hello.

Got the firts run in AP mode, and was able to set the wifi settings according to my AP.
Then it reboots as expected.
This first reboot was very long... Then I was able to login and get the main page.
It suddendly comes not responsive and I see that the Pi is back to OFF state.

I tried to power-on again, and boot process is very very long (>30 sec. with V3.1) > 90Sec with V4.10
I'm able to get main page, and only help page then shutdown again.

It is a Pi Zero 2W (512 Mo RAM - Wifi - BT) - No Red led, only green matching Pi state.

Got logs via Linux computer were I could read SD-CARD partition.

Will try another brand new SD-CARD (bought 2 samples) and let you know.

SD-CARD is a 16 Gb, brand new one, SAMSUNG SDHS EVO plus Class 10.

autourdupc, 2024-08-24:
Well.

New try with new SD-CARD -> Same thing.

Got AP mode and was able to enter my wifi settings, and also my login / password.

`running script wificredentials.pl running script wificredentials.pl sudo /usr/lib/cgi-bin/ecu/wifi/credentials_save.sh QraWifi ***MYwifiPassword***saving wifi credentials...
script nmcredentials_save.pl args: QraWifi ***MYwifiPassword***
\n
\n sudo /usr/lib/cgi-bin/ecu/wifi/setPasswd.cgi laurent **MyPassword**Content-type:text/html
running setPasswd.cgi
the command = htpasswd -bc /etc/apache2/.htpasswd laurent **MyPassword**
wificredentials.pl ready...
HTTP/1.1 200 OK
`
After reboot, got the main screen 
[image]

If I do nothing and wait for about 50 seconds, PI stops (power off)

I try to power on again, got the main screen 
Was able to get Info menu
[image]

PI stops after 50 sec.

Another try permits to get SYS menu
[image]

And it stops

autourdupc, 2024-08-24:
Also need to know the ssh access : root /mies do not login !

autourdupc, 2024-08-24:
And still have loopback issue...
running script sendZigbee.pl...
argument is 2710AABBCC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html
running testZigbee.cgi version 2024-jan-12
script testZigbee running with arg 2710AABBCC
sendZigbee write FE032710AABBCCDA nothing on serial0 after 2000 milisec
readZigbee
inMessage = no answer
command processed
HTTP:1/1 200 OK

autourdupc, 2024-08-24:
And finally, after restoring settings, it takes the version number of previous backuped data (it was V4.1 and now showed 3.1)...
[image]

and also 
[image]

patience4711, 2024-08-30:
I see that you didn't expand your filesystem. That is the first step to do after first boot-up (please see wiki). If you don't do this, there is not enough disk space for the system which can lead to malfunction. 

In the system page you probably cannot see the button for system expansion. This is because your premature backup restore.  If you ssh you can do: 
- change the content in the file /var/www/ecu-settings/expand.txt to 'n' save and visit (refresh) the page 'system'
- or
- type 'sudo raspi-config' and expand the filesystem. 
 
Then reboot. Normally the zigbee system should be up now.

"After restoring settings", i assume that you mean restoring your backed-up database? Indeed there is a problem here as this backup is a backup of the databases and the settings. So when you restore your backup, you get the version of your backup which is 3_10.  This is is a shortcoming in the total concept of the software.
Sadly the only thing i can do about this, is issue a software update to correct this. I'l do that soon. But before that i need to be sure that there are no other issues.

At boot the system tries to start the zigbee coordinator. If this failes it will try again every night at 4hr. I don't know why in your case it didn't start but when you can start it via console there is nothing wrong. Can you confirm that your system consequently refuses to start the zigbee at boot?

autourdupc, 2024-08-30:
Hello.

Taking your points...
I see that you didn't expand your filesystem. (or do you use a 4Gb sdcard) That is the first and important step to do after first boot-up (please see wiki). If you don't do this, there is not enough disk space for the system which can lead to malfunction.

-> According to this picture, the file system is enought to work properly... Right ?
This is the state of the disk after first boot.
[image]

After restoring the backup, this file systems indicates this (witch is wrong !!!)
[image]

In the system page you probably cannot see the button for system expansion.
-> False... I could see it !

"After restoring settings", i assume that you mean restoring your backed-up database?
-> Sure... restoring the database previously backuped on the 3.10 system.

So when you restore your backup, you get the version of your backup which is 3_10.
-> Yes... This is so strange... One would expect getting the software version (4_10), not the backup version (3_10)

 I'l do that soon. But before that i need to be sure that there are no other issues.
 -> Of course !

Can you confirm that your system consequently refuses to start the zigbee at boot?
-> No ! Coordinator start at boot as expected. But it consume all CPU ressources (at first start). Next boots are ok.

And finally, this version (4_10) is still not good for loopback tests...
```
running script sendZigbee.pl...
argument is 2710AABBCC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html
running testZigbee.cgi version 2024-jan-12
script testZigbee running with arg 2710AABBCC
sendZigbee write FE032710AABBCCDA nothing on serial0 after 2000 milisec
readZigbee
inMessage = no answer
command processed
HTTP:1/1 200 OK

```
-> This command ruined the system

Even after a reboot, it was not better.
I had to manually start coordinator to get a working system.

patience4711, 2024-08-30:
After first boot, your filesystem is surely less then 4 gb (the size of the image). So this cannot be correct. These figures are gathered with normal raspberry system commands so if it is wrong, i cannot help it.  After you expand your fs can be 16G

So when you restore your backup, you get the version of your backup which is 3_10.
-> Yes... This is so strange... And the disk details of the 3_10. One would expect getting the software version (4_10), not the backup version (3_10)
-> No.. not strange, i explained why this happens.

-> This command ruined the system. You issue 2710AABBCC and then....  what means ruined the system? 

I see however that the checksum is not the same as mine, (i have E9) looks like we have an old problem back.
I have to investigate what happend here.

patience4711, 2024-08-31:
Oke, but it is still not clear to me whether you expanded the fs or not? If you see that button you didn't.

About the loopbackcommand, in the new linux i made a library that contains all functions that have to do with the io to the zigbee module.  So polling, pairing, startcoordinator, etc all use this library and they seem to work. The custom zigbee command works perfectly in my system.

All the insights we have acquired at the time of the problems with the loopback command have been included in this library.  (the bufsize of sln and the crc calculation).  

I don't know what the problem is now, and i can't experiment because i dont have a rpizero2.

There is no separate slen calculation function anymore, this is included in the function sendZigbee().  Could that be the cause?

So my suggestion is that you examen this library yourself, maybe you see the problem. Could have to do with the buffersize of byteSend,  the slen line, id don't know. 
Below you'l see the crc function from the file /usr/lib/cgi-bin/ecu/libraries/serial_IO.h
```
 // ********* calculate the checksum ****************
char *checkSum(char Command[])
{
    char* buffer_1 = new char[254];
    char buffer_2[254] = {0};
    strncpy(buffer_1, Command, 2); // starting point the first 2 chars

    for (uint8_t i = 1; i <= (strlen(Command) / 2 - 1); i++)
    {
        strncpy(buffer_2, Command + i * 2, 2);
        sprintf(buffer_1, "%02X", StrToHex(buffer_1) ^ StrToHex(buffer_2));
    }
    return buffer_1;
}
```

```
char* sLen(char Command[])
{
    char* bufferSln = new char[254];
    //so if command =2710AABBCC then strlen=10 /2 =5 -2 = 3
    sprintf(bufferSln, "%02X", (strlen(Command) / 2 - 2));
    //cout << "slen = " << bufferSln << "\n<br>" << endl;
    return bufferSln;
}
```
After the weekend i will do some tests with bringing back the old separte sLen function.

autourdupc, 2024-09-01:
Hello.

Concerning the value of strlen(Command), are you sure you get each time an even value ?
Because if you get an odd value, what about the result of (strlen(Command) / 2 - 2) ?
strlen(Command) / 2 for an odd value will give a shorter value than expected or ?

patience4711, 2024-09-02:
The problem here is that when you restore settings, also the disk-size info (which is written at boot) and the version info is overwritten. I have to locate these files in another place to avoid that in future.

In my system there is never a problem with the custom zigbee command.  I compilled a new testZigbeetest.cgi [here](https://github.com/patience4711/RPI-APS-inverters/blob/main/testZigbeetest.cgi) for you to test.
If you copy that into /usr/lib/cgi-bin/ecu you can run it "./testZigbeetest.cgi 1270AABBCC" , it only simulates sending the message and prints what it is doing. So it won't ruin anything.
The checkSum should be E9

autourdupc, 2024-09-02:
Hey !

ecuuser seem not having rights to put the file into this folder... !
I use WinSCP... Need to fight to put it in the rightfolder.

```
ecuuser@192.168.20.114:/usr/lib/cgi-bin/ecu$ ./testZigbeetest.cgi 2710AABBCC
./testZigbeetest.cgi: line 7: syntax error near unexpected token `newline'
./testZigbeetest.cgi: line 7: `<!DOCTYPE html>'
ecuuser@192.168.20.114:/usr/lib/cgi-bin/ecu$
```

patience4711, 2024-09-02:
Maybe the better. I removed the file (too late it seems) because it sends a single "FE" to the coordinator which makes it crash. 
The right file is [here](https://github.com/patience4711/RPI-APS-inverters/blob/main/testZigbeetest.cgi)
To get root permissions you should "sudo su password"
I think your terminal is adding a cariage return and/or newline character, you can put that off.
[image]

autourdupc, 2024-09-02:
Hum...
This is the new version...
[image]

I use putty for ssh. I never got any issue regarding CR or LF... 

```
ecuuser@192.168.20.114:/usr/lib/cgi-bin/ecu$ ./testZigbeetest.cgi 2710AABBCC
./testZigbeetest.cgi: line 7: syntax error near unexpected token `newline'
./testZigbeetest.cgi: line 7: `<!DOCTYPE html>'

```

patience4711, 2024-09-02:
And about expanding the fs or not:
The fs info is written once the fs is expanded. So if you don't do that, this info will be incorrect. Via a database restore you inherit this value from your old system (which could be right or wrong).
If you not expand, you could run into troubles as your database is growing, logfiles growing, upload of software uprade tarballs, databasebackups etc. The minimum size is (on the save side) 8 Gb.

I made as softwareupdate [RPI-ECU-v4_2](https://github.com/patience4711/RPI-APS-inverters/blob/main/ecu_swupdate-v4_2.tar)
This relocates the infofiles for the version and the fs expand, so that this info is not overwritten by a database restore.
Furthermore i changed the testZigbee.cgi so that it uses the sLen function again, so it should work now on your system.
I just tested the following

- downloaded and wrote the imagefile.
- expanded the filesystem
- installed the swupdate to 4_2
- now the expandbutton appears so expanded again.
- after 2 reboots the filesystem info is correct (this was always true, never found out why)
- restored a database backup from version 3_9
- system info did not change and version still 4_2
- now tried the loopback command, works.

If you just install the upgrade (without expanding) you will get wrong sdcard info but that should be correct after a reboot. Also the expand button will appear, you can only get that away by clicking it and expand the fs system.

autourdupc, 2024-09-03:
Hello.

Just FYI with 4_1 version
[image]

-> Sudo is not better.

I will update du 4_2 and let you know.

Concerning filesystems infos, can we imagine a cron that updates one a day the filesizeinfo ? It could be cool to get the right value on the page.
Imagine a system running during one year without rebooting... Filesystems info will never update !

patience4711, 2024-09-03:
I was already experimenting with a script that runs every time you open the info page. To prevent wear of the sd-card this should be written in ramdisk. This didnt work, so im tryint to integrate in in the startCoordinator script that runs every night and at boot.

autourdupc, 2024-09-03:
I did some calculations about quantity of writes on an sd card.
My Freebox running on a sd-card write all on sd... 
Due to nb cycles available>1000 and wear liveling on sd cards, it can support up to 10 years writing everydays

autourdupc, 2024-09-04:
Hello.

I did the 4_2a update, without rebooting.

Got this now...
[image]

patience4711, 2024-09-04:
Oke so the loobeck issue is gone.

The spinner is shown when the chart is loading. It is not in the middle because you have to reload the stylesheet (remove browser settings and refresh the page).

autourdupc, 2024-09-05:
Hello.

This morning, the "SYS" menu displays all the data !
[image]

It seems that it works correctly an update once a day the SD-CARD occupation... Nice job !

patience4711, 2024-09-05:
That is already outdated haha. I found a way to get the info page to retrieve real-time info about the disk space. No more file writing/reading and always up-to-date. Available within the next update... If you can't wait then, in the file /var/www/html/protected/info.php you can "replace the part disk info (between the bold lines) wit the code below.

**". $vrij . "\<br>"\";**

```
// *********************   disk info ******************
$disk = shell_exec("df -h | grep /dev/mmcblk0p2");
echo "<br>current sd-card usage:<br> ";
echo"<table><tr><td>filesystem<td>size <td>used <td>available <td>use %";
$items = explode(" ", $disk);
$removed = array_pop($items); // remove the last element
echo "</td><tr>";
// print the array items, skip when empty
foreach($items as $v){
if ($v != '') { echo "<td> ". $v . "</td>";}
}
echo "</table>";
echo "<br>";
// **************  end disk info **********************
```

**$filename = '/var/www/ecu_data/startstop.txt';**

and remove this line or comment it (no help button needed anymore
``

Btw the available space is coincidently. When i issue an image of 16Gb, i shrink it as much as possible to just a little of free space, for obvious reasons. This time you got lucky... I would advice to monitor the free space, then you still can expand before you run into problems. I dont know how large the database can grow, especially with multiple inverters.

autourdupc, 2024-09-05:
Installed 4_2b
-> "b" do not appear in the installed version. Only 4_2 !
[image]

Also tested loopback -> Great !
[image]

Rest seem working as expected !

patience4711, 2024-09-05:
Could you replace the memory part in info.php to se if it fits your needs?

// *********************   memory ******************
$mem = shell_exec("free --mega | grep Mem");
$memitems = explode(" ", $mem);
echo"<table><tr><td style='width:190px;'>Memory usage<td>Total<td>used <td>Free</td></tr><tr><td>Mb's";
$hits = 0;
foreach($memitems as $v){
if ($v != '' && $v != 'Mem:' && $hits <3) { 
   echo "<td> ". $v . "</td>";
   $hits += 1;
   }
}
echo "</table>";
echo "<br>";

patience4711, 2024-09-05:
Oke, nobody never noticed that there was something wrong with that. Apparently nobody uses it. I always trusted the method was oke while it uses a linux system command. This method extracts the values in another way. 
This change is reflected in update v4_2c and yes the version stays v4_2.

autourdupc, 2024-09-05:
> If you can't wait then 

I did... Its perfect !
[image]

autourdupc, 2024-09-05:
> Apparently nobody uses it

Not my case... If there is a functionnality available, I try it :)
This is why I find so many bugs !

patience4711, 2024-09-05:
If only you find them, i wouldn't call it bugs haha. But i am happy with your approach, it really helps to improve the system.

autourdupc, 2024-09-05:
> If only you find them, i wouldn't call it bugs haha. But i am happy with your approach, it really helps to improve the system

Its your fault... You told me to expand... So I focused on storage... So info page, so incorrects values... ah ah !

Thank you very much for the job, and your "patience", as always !

autourdupc, 2024-09-05:
Another issue in details.php...

I can see a line of the table :
`echo "<tr><td title='based on 230g/KWh' class='nm' >CO<sub>2</sub> emission saved: <td id='ch0' title='based on 230g/KWh'>" .number_format($total*0.230, 1) . " kg";`

But not visible on the output
[image]