patience4711/RPI-APS-inverters #125: Showing power graph for past days not working

ElTh0r0, 2024-09-04:
Hi,
first of all: Many thanks for the great work you are providing with this project!

Yesterday I migrated to v4.x on my Raspberry pi Zero (burned v4.1 onto microSD, added to wifi, expand FS, updated to v4.2, cleaned DB, restored DB backup). Everything works except showing power graph for any day in the past (showing the current graph of the day works). When selecting e.g. Sep. 2nd in the calender, the "spinning wheel" is shown for few seconds as it would collect data from the DB, but afterwards the calender changes back to today's date and today's graph. I cleared the browser cache and tried different browsers on desktop & mobile; behavior is the same. (unfortunately I have only one microSD and no possibility to double check back on v3.x, but I am quite sure this was working)

The other graphs work; e.g. energy graph when selecting previous month has no issues. The old energy data seems to be available inside the DB, as I can query the old data from the DB page (see data below).

Can somebody reproduce the issue? Is there something I can check myself / any particular files I could dig into?

```
running script dbQuery.pl...
arguments are invData and 0 and 2024-09-02
** invData query all values in date 2024-09-02 **
curl -G http://localhost:8086/query?db=invData --data-urlencode "q=select * from inv0 WHERE time > '2024-09-02' AND time < '2024-09-02'+1d ORDER BY desc"

2024-09-02T18:10:07.283263335Z e= 0 p= 0
2024-09-02T18:00:09.348612901Z e= 0.1 p= 1.5
2024-09-02T17:55:08.522537935Z e= 0.4 p= 4.2
2024-09-02T17:50:08.296537025Z e= 0.5 p= 5.9
2024-09-02T17:45:08.305893399Z e= 0.6 p= 6.9
2024-09-02T17:40:08.32042619Z e= 0.8 p= 9.8
2024-09-02T17:35:08.346611494Z e= 1.2 p= 13.9
2024-09-02T17:30:08.360725819Z e= 1.4 p= 17.3
2024-09-02T17:25:11.571794121Z e= 1.7 p= 19.9
2024-09-02T17:20:08.467603307Z e= 1.9 p= 22.2
2024-09-02T17:15:08.356213901Z e= 2 p= 23.7
2024-09-02T17:10:08.430158197Z e= 2.2 p= 26.7
2024-09-02T17:05:08.526729601Z e= 2.3 p= 27.1
2024-09-02T17:00:08.63825106Z e= 1.1 p= 29.3
2024-09-02T16:55:08.331977941Z e= 2.9 p= 34.8
2024-09-02T16:50:10.554289684Z e= 2.9 p= 34.2
2024-09-02T16:45:08.351385917Z e= 2.6 p= 30.7
2024-09-02T16:40:08.381639536Z e= 2.8 p= 34
2024-09-02T16:35:08.43408165Z e= 4.1 p= 48.9
2024-09-02T16:30:08.312115789Z e= 5.1 p= 61.1
2024-09-02T16:25:10.565168127Z e= 4.6 p= 54.6
2024-09-02T16:20:08.379822082Z e= 7.5 p= 89.6
2024-09-02T16:15:08.378053111Z e= 10.3 p= 124.1
2024-09-02T16:10:08.800884582Z e= 13 p= 156
2024-09-02T16:05:08.339697727Z e= 13.4 p= 160.8
2024-09-02T16:00:08.469093269Z e= 11.2 p= 135
2024-09-02T15:55:08.385710059Z e= 13.5 p= 161.5
2024-09-02T15:50:08.329805294Z e= 17.2 p= 208.4
2024-09-02T15:45:11.302162186Z e= 22.8 p= 270.6
2024-09-02T15:40:08.329282419Z e= 25.3 p= 303.8
2024-09-02T15:35:08.361287721Z e= 27.1 p= 325.1
2024-09-02T15:30:08.33308968Z e= 28.6 p= 343.5
2024-09-02T15:25:08.355448742Z e= 28 p= 335.9
2024-09-02T15:20:08.36955763Z e= 28.6 p= 343.9
2024-09-02T15:15:08.519560155Z e= 30 p= 358.5
2024-09-02T15:10:08.327149834Z e= 31 p= 374.7
2024-09-02T15:05:10.480220468Z e= 32.7 p= 391.7
2024-09-02T15:00:08.403318125Z e= 30.7 p= 368.2
2024-09-02T14:55:08.378517457Z e= 28.3 p= 339.7
2024-09-02T14:50:08.84558532Z e= 31.6 p= 381.3
2024-09-02T14:45:11.026656156Z e= 35.6 p= 424
2024-09-02T14:40:08.489060021Z e= 35.6 p= 427.2
2024-09-02T14:35:08.360022243Z e= 36.4 p= 436.4
2024-09-02T14:30:08.37014175Z e= 36.3 p= 435.3
2024-09-02T14:25:08.48438301Z e= 25.1 p= 301.4
2024-09-02T14:20:08.316441707Z e= 28.3 p= 342.4
2024-09-02T14:15:10.574242095Z e= 22.8 p= 271.4
2024-09-02T14:10:08.328498912Z e= 18.7 p= 224
2024-09-02T14:05:08.389524315Z e= 19.6 p= 235.3
2024-09-02T14:00:08.348601502Z e= 16.5 p= 199.8
2024-09-02T13:55:11.257537491Z e= 20.6 p= 245.8
2024-09-02T13:50:10.603722766Z e= 30.8 p= 369.2
2024-09-02T13:45:10.770223352Z e= 42 p= 504.1
2024-09-02T13:40:09.82287184Z e= 44.1 p= 525.6
2024-09-02T13:35:08.31927711Z e= 47.6 p= 571.5
2024-09-02T13:30:08.463590286Z e= 38.8 p= 465.4
2024-09-02T13:25:08.392432644Z e= 27.1 p= 325.2
2024-09-02T13:20:08.491553649Z e= 21.1 p= 253.4
2024-09-02T13:15:08.379120819Z e= 18.4 p= 221.8
2024-09-02T13:10:09.368507765Z e= 16.9 p= 202.4
2024-09-02T13:05:09.045124277Z e= 17.2 p= 206.1
2024-09-02T13:00:08.401685957Z e= 17.3 p= 207.9
2024-09-02T12:55:08.374851755Z e= 13.2 p= 159.6
2024-09-02T12:50:09.861313414Z e= 12.5 p= 148.5
2024-09-02T12:45:08.413868548Z e= 10.8 p= 131.3
2024-09-02T12:40:12.933353191Z e= 9.1 p= 107.4
2024-09-02T12:35:08.349874998Z e= 7.6 p= 92.1
2024-09-02T12:30:10.603695997Z e= 7 p= 82.6
2024-09-02T12:25:08.339771679Z e= 6.1 p= 73.7
2024-09-02T12:20:08.347163951Z e= 5.9 p= 70.8
2024-09-02T12:15:08.384102921Z e= 6 p= 72.3
2024-09-02T12:10:08.627069851Z e= 6.5 p= 78.3
2024-09-02T12:05:08.325278578Z e= 7 p= 83.9
2024-09-02T12:00:08.462925403Z e= 5.8 p= 69
2024-09-02T11:55:08.351730731Z e= 5.5 p= 65.6
2024-09-02T11:50:08.436158292Z e= 5.7 p= 68.4
2024-09-02T11:45:08.566297457Z e= 5.9 p= 70.3
2024-09-02T11:40:08.434388816Z e= 6.2 p= 73.8
2024-09-02T11:35:08.354742269Z e= 7.4 p= 88.6
2024-09-02T11:30:09.81441307Z e= 14.6 p= 174.6
2024-09-02T11:25:09.009866933Z e= 18.6 p= 222.9
2024-09-02T11:20:08.414314291Z e= 11.6 p= 138.7
2024-09-02T11:15:08.381664758Z e= 8.4 p= 101.4
2024-09-02T11:10:10.563120846Z e= 29.7 p= 177.5
2024-09-02T11:00:08.420650062Z e= 35.7 p= 428.8
2024-09-02T10:55:08.310661414Z e= 36.9 p= 443.2
2024-09-02T10:50:08.337782569Z e= 23.2 p= 278.5
2024-09-02T10:45:08.381025865Z e= 37.5 p= 450.5
2024-09-02T10:40:08.484967771Z e= 19.1 p= 229.3
2024-09-02T10:35:08.33514275Z e= 16.3 p= 197.9
2024-09-02T10:30:12.374683208Z e= 19 p= 225
2024-09-02T10:25:08.466342513Z e= 39.4 p= 476.4
2024-09-02T10:20:10.611988776Z e= 45.3 p= 543.4
2024-09-02T10:15:10.005931721Z e= 42.6 p= 507.7
2024-09-02T10:10:08.401927786Z e= 24.4 p= 295.3
2024-09-02T10:05:11.26677892Z e= 21.6 p= 256.3
2024-09-02T10:00:08.355732565Z e= 32.8 p= 393.3
2024-09-02T09:55:08.374724912Z e= 42.4 p= 509
2024-09-02T09:50:08.418116644Z e= 42.4 p= 508.4
2024-09-02T09:45:08.354069646Z e= 43.5 p= 522.4
2024-09-02T09:40:08.365013736Z e= 75.4 p= 454.8
2024-09-02T09:30:10.917511572Z e= 39 p= 463.3
2024-09-02T09:25:08.350032384Z e= 35.4 p= 426
2024-09-02T09:20:08.530487364Z e= 36.5 p= 437.4
2024-09-02T09:15:08.436254532Z e= 34.9 p= 418.9
2024-09-02T09:10:08.446042234Z e= 35.1 p= 420.6
2024-09-02T09:05:08.452860599Z e= 33.6 p= 405.7
2024-09-02T09:00:10.566427016Z e= 34.3 p= 409.4
2024-09-02T08:55:08.395239684Z e= 32.9 p= 394.6
2024-09-02T08:50:08.348726017Z e= 32.3 p= 387.6
2024-09-02T08:45:08.398221417Z e= 32.6 p= 391
2024-09-02T08:40:08.373935989Z e= 31.7 p= 382.5
2024-09-02T08:35:10.626323577Z e= 31.1 p= 373.7
2024-09-02T08:30:10.497835371Z e= 29.8 p= 355.8
2024-09-02T08:25:08.331696455Z e= 29.1 p= 349.8
2024-09-02T08:20:08.924505257Z e= 29 p= 347.4
2024-09-02T08:15:08.307708457Z e= 28.1 p= 337.3
2024-09-02T08:10:08.366341501Z e= 28.6 p= 343.7
2024-09-02T08:05:08.363720816Z e= 28.6 p= 343.1
2024-09-02T08:00:08.511541391Z e= 27.6 p= 331.4
2024-09-02T07:55:08.34910868Z e= 26.7 p= 320.4
2024-09-02T07:50:08.413117114Z e= 25.6 p= 307.4
2024-09-02T07:45:08.487045177Z e= 24.1 p= 289.1
2024-09-02T07:40:08.346229281Z e= 22.7 p= 272.4
2024-09-02T07:35:08.408612934Z e= 20.6 p= 248.1
2024-09-02T07:30:09.17007706Z e= 19.5 p= 233.3
2024-09-02T07:25:08.318187602Z e= 18.6 p= 223.3
2024-09-02T07:20:08.330441887Z e= 17.5 p= 209.7
2024-09-02T07:15:08.39737168Z e= 16.3 p= 196.7
2024-09-02T07:10:09.82686048Z e= 15.5 p= 184.4
2024-09-02T07:05:08.428887214Z e= 14.3 p= 171.4
2024-09-02T07:00:08.328796971Z e= 13.1 p= 159.1
2024-09-02T06:55:11.064392662Z e= 12.4 p= 147.3
2024-09-02T06:50:08.335881674Z e= 11.3 p= 136
2024-09-02T06:45:09.923850279Z e= 10.2 p= 122.8
2024-09-02T06:40:09.801983431Z e= 9.2 p= 111.4
2024-09-02T06:35:12.377986902Z e= 8.4 p= 99.7
2024-09-02T06:30:08.379001516Z e= 7.2 p= 87
2024-09-02T06:25:08.589068029Z e= 6.4 p= 76.4
2024-09-02T06:20:08.332533351Z e= 5.7 p= 68.5
2024-09-02T06:15:08.353914748Z e= 5 p= 59.7
2024-09-02T06:10:08.424805282Z e= 4.3 p= 51.3
2024-09-02T06:05:08.381853331Z e= 3.8 p= 45
2024-09-02T06:00:08.286886846Z e= 3.2 p= 38.5
2024-09-02T05:55:08.376831529Z e= 2.7 p= 32.1
2024-09-02T05:50:08.314039203Z e= 2.2 p= 26.4
2024-09-02T05:45:08.4807874Z e= 1.7 p= 21
2024-09-02T05:40:08.338302671Z e= 1.4 p= 17.2
2024-09-02T05:35:08.317636487Z e= 1.1 p= 12.9
2024-09-02T05:30:08.38917247Z e= 0.8 p= 10
2024-09-02T05:25:08.353669836Z e= 0.8 p= 9.1
2024-09-02T05:20:08.480708727Z e= 0.5 p= 5.8
2024-09-02T05:15:08.359821223Z e= 0.4 p= 4.7
2024-09-02T05:10:08.350796429Z e= 0.1 p= 1.2
2024-09-02T05:05:08.44892044Z e= 0 p= 0.2
2024-09-02T05:00:08.332028981Z e= 0 p= 0
2024-09-02T04:40:07.643602898Z e= 0 p= 0

dbCmd command ready
HTTP:1/1 200 OK
```

patience4711, 2024-09-04:
Sorry to hear that. If your database is oke we can solve this.  So you say you can query a date in the past via the databaseconsole? Then your database is oke, that's reassuring.
I tried to repoduce this and restored a backup from my working system to my spare system. No problems found so far. The file /var/www/ecu_data/chartDate.txt exists. When i remove it, i have the same behaviour as you. 

The directory /var/www/ecu_data is replaced by the database backup so it seems that the backup misses that file.
If you can ssh into the system and make this file and give it the right owner and permissions, it should work again.

ElTh0r0, 2024-09-04:
> I tried to repoduce this and restored a backup from my working system to my spare system. No problems found so far. The file /var/www/ecu_data/chartDate.txt exists.

I checked my latest backup from yesterday and another old backup from 2023 - none is including `chartDate.txt`. So not sure, if something is "wrong" with that file on my system since beginning?! Additionally I just checked the v3.3 and v4.1 image and the file isn't included there by default either? Shouldn't the system create the file, if it doesn't exist? Where should the file come from, if one is starting with a fresh burned image (when no backup exists to be restored)?

> The directory /var/www/ecu_data is replaced by the database backup so it seems that the backup misses that file.
If you can ssh into the system and make this file and give it the right owner and permissions, it should work again.

I tried. I created an empty file `chartDate.txt` and changed owner permission to the same as the other files (`-rwxr--r-- 1 www-data www-data`), but still the same issue. Which owner & permission has your `chartDate.txt`?

I digged further: Browser dev tools are showing an error for 1sec:
[image]
So I checked the /var/log/apache2/error.log:
```
[Wed Sep 04 16:58:44.615508 2024] [cgid:error] [pid 12256:tid 3069732224] (13)Permission denied: AH01241: exec of '/usr/lib/cgi-bin/setchartDate.pl' failed
[Wed Sep 04 16:58:44.619469 2024] [cgid:error] [pid 6445:tid 3027223488] [client 192.168.178.20:58508] End of script output before headers: setchartDate.pl, referer: http://192.168.178.50/chartjs/chart_pe.php?page=0&inv=0
```
Do you have an idea which permission is wrong here / how to correct it?

ElTh0r0, 2024-09-04:
I just saw, that there are two `setchartDate.pl`. One in `/usr/lib/cgi-bin` with permission `-rwxr--r--` and one in `/usr/lib/cgi-bin/ecu` with permission `-rwxr-xr-x`. Second file in subfolder ecu has 'x' permission for execution, but not the file in `/usr/lib/cgi-bin`, which is triggered from the website!

```
root@192.168.178.50 /usr/lib/cgi-bin# ls -lh
total 16K
drwxr-xr-x 6 root root 4.0K Sep  3 15:03 ecu
-rwxr--r-- 1 root root  559 Mar 13 08:24 setchartDate.pl
-rwxr-xr-x 1 root root 3.7K Mar 13 16:06 swUpgrade.pl
drwxr-xr-x 2 root root 4.0K Mar 13 16:45 test
```

ecu subfolder:
```
root@192.168.178.50 /usr/lib/cgi-bin/ecu# ls -lh
total 2.2M
-rwxr-xr-x 1 root root 6.2K Sep  3 15:03 autoPoll.pl
[...]
-rwxr-xr-x 1 root root  559 Sep  3 15:03 setchartDate.pl
[...]
```

So I did `chmod +x setchartDate.pl` for `/usr/lib/cgi-bin/setchartDate.pl`  and the graph generation works again!
Where something went wrong here, I cannot evaluate - maybe you can? I can only wild guess, that first file without 'x' persmission was placed in '/usr/lib/cgi-bin/' by accident and the php files accidently calling this file instead of the one in 'ecu' subfolder?

patience4711, 2024-09-04:
Correct. The wrong script was called, in my system it works because it has enough permissions but when tarred the seem to change.  The problem is in the page /var/www/html/chartjs/chart_pe.php.
if you change the form declaration like so:
`echo '<form id="formulier" action="/cgi-bin/ecu/setchartDate.pl" target="hiddenFrame">';`
it will work.

ElTh0r0, 2024-09-04:
Many thanks for your quick response and providing a patch! I can confirm that after applying the patch, displaying the graph of the past day works.
I really appreciate your support! 🙂

closed 2024-09-04 by ElTh0r0