patience4711/RPI-APS-inverters #126: New stats for disk usage

autourdupc, 2024-10-17:
Hello Patience.

As I told you I wanted to play with RPI-ECU.

For a first time, I had to understand the code you did and remember languages from my side (I'm not working any more in software development).
  
I'm pleased to give you some files I created based on you good job. It permits for me to understand Influx database and Chartjs.

This extension to your job do:
1) Collect every 6 hours data disk usage and put them in a database
2) Using lbo_info in protected mode, you can display a dynamic info page refreshed every seconds (you can then see the CPU usage and temperature in real time)
3) Have charts to see disk usage variation for the last past 3 month 

I made 4 files that should be stored into their respectives paths
/var/www/html/chartjs/lbo_chart_disk.php
/var/www/html/chartjs/lbo_chartdata_disk.php
/var/www/html/protected/lbo_info.php
/usr/lib/cgi-bin/ecu/lbo_sysinfos.sh

Need to sudo chmod +x /usr/lib/cgi-bin/ecu/lbo_sysinfos.sh

Also need to create a cron job :
sudo crontab 0 */6 * * * /usr/lib/cgi-bin/ecu/lbo_sysinfos.sh
and launch manually via sudo /usr/lib/cgi-bin/ecu/lbo_sysinfos.sh

Replace in you menu the call to "info.php" by "lbo_info.php"

After 6 hours, you will have collected one more set of data for disk usage.

Here are the files
[lbo_rpi-ecu.zip](https://github.com/user-attachments/files/17420728/lbo_rpi-ecu.zip)

[image]

[image]

Have fun !

patience4711, 2024-10-18:
Thanks, you played a lot. Very nice. I am not going to implement this because i don't see the need for it. This is more like nice to have (as some other information too).  Maybe you can publish a fork.
Keep up the good work.

autourdupc, 2024-10-21:
Rhhhhaaaaaa !
I found this so... Funny !

I'll try to do better the next time.