patience4711/RPI-APS-inverters #127: YC600 with only 1 panel not reporting DC Volts and DC Amps

robertbeudeker, 2024-10-25:
First off all many thanks for this wonderful solution. I must say i'm impressed and very thankful. My ECU-R is defective. The ECU-R costs around 180 euro's and this solution is a fraction of the cost and maybe even better.

I have 5 inverters (YC600) with 9 solar panels. the 4 inverters with 2 panels connected all work just fine.
I do have a problem with the inverter that has only 1 panel connected. On this one, is seems that the DCV and DCA are swapped.
DC volts and DC Amp are not reporting correct data. (Or so low that i can not be the correct data)
Below is the json for this inverter at the moment.
you can see the first array element of pwr is 133 witch is correct, also "en" and enStack is correct. dcc channel 1 and 2 are swapped, also dcv channel 1 and 2 are swapped.
This same issue occures with the mqtt message
{ "pwr": [ 133.2, 0.0, 0.0, 0.0 ], "en": [ 49.9, 0.0, 0.0, 0.0 ], "totalenStack": 49.9, "dcc": [ 0.0, 4.9, 0.0, 0.0 ], "enStack": [ 49.9, 0.0, 0.0, 0.0 ], "dcv": [ 2.9, 32.6, 0.0, 0.0 ], "heath": 12.4, "freq": 50.0, "acv": 233.1, "totalpw": 133.2, "time": 9447, "totalen": 49.9, "idx": 2147483647 }

robertbeudeker, 2024-10-25:
forgot to mention, i'm running the 4.2 image.

patience4711, 2024-10-28:
Indeed it looks like dcv and dcc are swapped. It's not difficult to repair, i will issue a software update tomorrow.