patience4711/RPI-APS-inverters #129: New chart for power production for all inverters in a same graph

autourdupc, 2024-11-15:
Hello Patience.

Let me provide a new php file that displays power measurements for all inverters in one single chart.

It displays in real time the total power of all inverters, displays the max power of the day, the pool duration and the average power for the day (using sunset/sunrise).
One can change the date on the calendar to get chart for other days.
It refreshs every 112 secondes
It works on phone device and on computers browser.

[image]

Find below the php file named lbo_chart_p.php
Put it in the /var/www/html/chartjs folder ->  /var/www/html/chartjs/lbo_chart_p.php
 Need to sudo chmod +x /var/www/html/chartjs/lbo_chart_p.php

Then call it via http://your-ip/chartjs/lbo_chart_p.php

You will find the source code fully commented for maintenance and explanations.
I actually do not know how to link it to your existing main page.

Here is the file...
[lbo_chart_p.zip](https://github.com/user-attachments/files/17781755/lbo_chart_p.zip)

I hope to live up to your expectations.

Cheers,
Laurent.

defaliz, 2024-11-18:
Thank a lot Autourdupc, very useful
i'll try..
french ? bonne journée et merci encore...

autourdupc, 2024-11-18:
> Thank a lot Autourdupc, very useful i'll try.. french ? bonne journée et merci encore...
Oui :)

I also made a calculation of energy, but seem to be not the same as the one calculated by Patience, expecially on low values...
I don't know who got the truth... This is why this calculation is still not available in this chart... I need to see with Patience when he got time.

defaliz, 2024-11-18:
hi,
in the file there is a include setTimezone.php, but where is this file ?
regards

defaliz, 2024-11-18:
here is the warnings 
Warning: include(setTimezone.php): failed to open stream: No such file or directory in /var/www/html/chartjs/lbo_chart_p.php on line 34

Warning: include(): Failed opening 'setTimezone.php' for inclusion (include_path='.:/usr/share/php') in /var/www/html/chartjs/lbo_chart_p.php on line 34

autourdupc, 2024-11-18:
> in the file there is a include setTimezone.php, but where is this file ?

Hum... No warning here...
I run version RPI-ECU 4_2.
Anyway, I do not use the timezone, so you can comment both lines like this
```
// We need to set the timezone = ok
//include 'setTimezone.php';
//$tz = date_default_timezone_get();
```

defaliz, 2024-11-18:
Ok thanks..

autourdupc, 2024-11-18:
> Ok thanks..
Is it running well ?

defaliz, 2024-11-18:
Oups la version is RPI-ECU-v3_10, i need tobuodate...

autourdupc, 2024-11-18:
> , i need tobuodate..

DO not forget to backup database !!!

autourdupc, 2024-11-21:
@defaliz 
> Warning: include(setTimezone.php): failed to open stream: No such file or directory in /var/www/html/chartjs/lbo_chart_p.php on line 34

In fact, this file is mandory. If not present the timezone comes UTC so chart time values come bad.
Maybe the chart can be ok in previous version of RPI-ECU without this line but time sould be wrong.

In the 4.2 version, there is no warning and everything goes right with this thoses lines.

defaliz, 2024-11-21:
Ok i really need to update  my rpiEcu...

autourdupc, 2024-11-25:
Hello.
> wel the chart will pretty much look like the charts of the individual inverters so i thought it doesn't add much valuable
Grrrrr !
It depends on how the panels / inverters are oriented... If all are the same position, sure, it is the same chart... But if they are east/south/west, then the SUM is interesting !

Concerning power, I got it, but I find values not the same as yours... I asked Ai "Claude" to help me, and my calculation is okay according to Claude, so I need to do more tests.

Today, due to the storm all my 5 panels went away... Cables I made are still in good health, but thoses I bought are broken (the crimps have come loose).
I need to rebuild all the supports for the panels... Need some days !

autourdupc, 2024-11-30:
Well... I'm not lucky...
I've just reinstalled my solar panels, and found RPI-ECU working strangely...
Made a reboot and nothing !
Seem that the SD-CARD went corrupt.

It a SAMSUNG EVO...
I lost again my database

autourdupc, 2024-12-08:
Hello @patience4711 

I published a new release of the chart with energy calculation, but I found a big difference with your calculation.
I asked an AI (Claude) to help me resolve the issue but the AI gave me the same results.
I also gave the AI a full day measurements table and it gave the same calculation as me...

I'm really confused with the Energy values. What is your opinion about it ?

patience4711, 2024-12-13:
I don't see the problem. The energy values come from the inverter, it is the difference between a counters begin- (normally 0 at reboot, start of the day) and end-value. You don't have to calculate that.

I don't know how you calculate, maybe average current * average voltage or so? these values are not useable because they reflect only the level at polling time. 
If you poll a voltage of 40 and 5 min later another poll of 44 you can't say that the average voltage in this period is 42 because meanwhile the voltage could be 50 at some time. The same goes for the current. So you never can tell what the amount of energy was during the period between 2 polls, let alone during a whole day.

autourdupc, 2024-12-13:
Energy can be calculated...
It is the power given during a time period.
We can calculate the energy even if we have one measure each 5 minutes. Of course, we don't have the value between the 2 points, but the energy calculation is still good.
Try to read some websites, or ask AI systems on energy calculation, they ill give you the same answer.

As I told you, I tried to give the whole dataset for a full day to an AI, and it give me the same results as me...

Actually in my region we have fog, so the values are bad - aroud 40 W produced (with peak at 120 W) for 1200 W installed.
The power variation between 2 collected points will not vary very much... 
ANd we can also say that we could have same variation less than variation more... So in the average the SUM of variations tend to reach 0.

I definitively wonder why energy calculation is so different !

defaliz, 2024-12-13:
> My mistake... It is not fog but clouds... Yes, I'm in France, Lyon.Le 13

Same here in Normandy, no sun since many days...

patience4711, 2024-12-16:
Consider this graph. The black line are power values polled every 5 minutes. When we would poll every second, the power chart could look like the red line.
The energy would be the area of the graph in which case they are quite different. 

  [image]

autourdupc, 2024-12-16:
Sure, I understand, but usually, the values above and the values under are quite equals so the error is small...
And this is true when solar intensity varies in cloudy conditions with sun/clouds alternances.
Actually, all teh sky is cloudy, so there is a continuous power production without littles variations.

And finally, my calculation is so different of the inverter measurements... Approx 100% error !

patience4711, 2024-12-17:
Anyway your calculation can't be right, too much difference. I just stick to the values the inverter provides, there is no way i can prove that they are wrong.  

If your measurements are: 500W, 600W, 550W, 520W
Sum = 2170W
Consumption = 2170 × (1/12) = 180.83 Wh

Energy calculation is about power over a period of time, if you have a constant powerlevel of 2170 during 5 minutes, then the 180.83 Wh is correct. The closest approach is avarage power level during 5 min/12 but even then, if you don't know what happened during this time you can't calculate the energy.

autourdupc, 2024-12-17:
Hello @patience4711 

I'm pleased to tell you that I found my mistake in my code for energy calculation.

Now I'm able to get the same value as yours. V1.30 is now okay. Remains some minors enhancements I need to do

defaliz, 2025-01-07:
Hi autourdupc
i've test your last version 1.5 with last version de rpiecu and i've errors in the page, like that :

Fatal error: Uncaught InfluxDB\Client\Exception: error parsing query: found WHERE, expected identifier at line 1, char 28 in /usr/lib/cgi-bin/ecu/vendor/influxdb/influxdb-php/src/InfluxDB/ResultSet.php:49 Stack trace: #0 /usr/lib/cgi-bin/ecu/vendor/influxdb/influxdb-php/src/InfluxDB/ResultSet.php(39): InfluxDB\ResultSet->validate() #1 /usr/lib/cgi-bin/ecu/vendor/influxdb/influxdb-php/src/InfluxDB/Driver/Guzzle.php(107):

autourdupc, 2025-01-09:
@defaliz 
This is very strange... 
Please give copy of version from http://your_ip/details.php

Then
sudo ls -l /var/www/html/chartjs/
should provide something like
-rwxr-xr-x 1 root root  16640 Dec 28 12:13 lbo_chart_p.php
-> Give me yours

Then
sudo cat /var/www/ecu_data/inverterCount.txt
should provide something like
2
-> Give me yours

sudo ls -l /var/www/ecu_data/inverters/
should provide something like 
-rw-r--r-- 1 www-data www-data 427 Dec  7  2021 invData0
-rw-r--r-- 1 www-data www-data 169 Jul 13 10:04 invProperties0
-rw-r--r-- 1 www-data www-data  96 Dec  6 21:35 invProperties1
-> Give me yours

defaliz, 2025-01-10:
Hi,
yes very strange, il's working with rpiecu 3_10
here is infos...
details.php

software version	RPI-ECU-v4_3
number of inverters	1
number of panels	2
polling starts at	08:46 hr
polling stops at	17:19 hr
overall total energy	2.0 kWh
CO2 emission saved: 	0.5 kg
trees planted: 	0.0

-rwxr-xr-x 1 root root 16697 Jan  7 11:57 /var/www/html/chartjs/lbo_chart_p.php

 cat /var/www/ecu_data/inverterCount.txt --> 1 pour l'instant 1 seul inverter

 ls -l /var/www/ecu_data/inverters/

-rw-r--r-- 1 www-data www-data 427 Dec  7  2021 invData0
-rw-r--r-- 1 www-data www-data 427 Dec  7  2021 invData0.sav
-rw-r--r-- 1 www-data www-data  98 Sep 22  2022 invProperties0.sav
-rw-r--r-- 1 www-data www-data 177 Apr 21  2023 invProperties0_sav
-rw-r--r-- 1 www-data www-data 174 Jan  9 12:08 invProperties1
-rw-r--r-- 1 www-data www-data 105 Apr 21  2023 invProperties1_sav

defaliz, 2025-01-10:
i found the error
variable $from is empty...because $InverterCount = 0
if i try $InverterCount = 1 so all works fine...
inverterCount.txt contain 1

actually i've only one inverter because the other is out of order, i think it's in the loop
for ($i=0 ; $i < $InverterCount ; $i++)
    {
    $from = $from."inv".$i.",";
    }

autourdupc, 2025-01-11:
Hello @defaliz 

Can you please try this verion ?
[lbo_chart_p.zip](https://github.com/user-attachments/files/18386084/lbo_chart_p.zip)

autourdupc, 2025-01-11:
Hello @defaliz 

Can you please test this 1.6 version ?
If working I will publish.
[lbo_chart_p.zip](https://github.com/user-attachments/files/18386102/lbo_chart_p.zip)

autourdupc, 2025-01-11:
@defaliz If you are online, tell me now

defaliz, 2025-01-11:
Hi
works perfectly ! (just change number version and it's OK)
thanks for quick correction...
do you have succeeded with mosqitto and HA ?

[image]

defaliz, 2025-01-11:
Ok, i've downloaded last version 1.6, all is good now
thanks and have a good evening

defaliz, 2025-01-14:
Each graph of power production finished at 13h...

defaliz, 2025-01-16:
another very strange thing , today production graph chartjs/chart_pe.php?page=0&inv=1
on arrow production = 92,8 w

[image]

and on your graph, arrow production is the same 92,8 w but not at the same time...

[image]

defaliz, 2025-01-17:
Hi,
yes gps settings are correct, all the other graphs are good... and start and end polling too...
as you can see, this part of your graph have right time but it is the start of the day and it's placed after the part of the afternoon...

[image]

defaliz, 2025-01-17:
merci, c'est fait, j'ai envoyé un message via le site

autourdupc, 2025-01-23:
New version 1.70...

defaliz, 2025-01-23:
Merci 👍

EricMicPro, 2025-05-23:
Hello All,

First, many thanks to @autourdupc for this chart.

If like me it is boring you a bit to type the address in your Web browser, you can change in the "index.php" file, the following line:
    echo "<tr><td class='nm'>totals<td class='hihi'>" . number_format($total_pw, 1, '.', '') . "<td class='haha'>" . number_format($total_en, 2, '.', '');

by this one:
    echo "<tr><td class='nm'>totals<td class='c2'><a href='chartjs/lbo_chart_p.php'><button class='linkbt'>" . number_format($total_pw, 1, '.', '') . "<td class='haha'>" . number_format($total_en, 2, '.', '');

This will add a button on the Totals line for Power W column.

defaliz, 2026-09-03:
hi @autourdupc 
i've a problem with the last file lbo_chart_p.php
it' ok on a smartphone screen but on a PC display is to narrow...
how could i correct this ?
regards

[image]

autourdupc, 2026-09-03:
@defaliz 

Very strange... I use Chrome and it works...
It works also in Firefox, and in Edge !

[image]

defaliz, 2026-09-03:
What's strange is that when the page loads, you can clearly see the scaling down.
i use rpiecu_v5_6 (with new feature throttling) and i've try 2 others browers, chrome and brave, it's the same...
not a huge problem... i'll survive...

autourdupc, 2026-09-03:
Hum... Maybe a stylesheet change in RPI-ECU... I need to check, but not planned soon !

autourdupc, 2026-09-03:
@defaliz 

Did you update or did you reinstall from new installation ?

autourdupc, 2026-09-03:
@defaliz 
Well.

Try this :
1) Download the zip file
[lbo_stylesheets.zip](https://github.com/user-attachments/files/31799278/lbo_stylesheets.zip)
2) Extract the 2 files from it
3) Put them in "var\www\html\"
4) Modify lbo_chart_p at lines 306 and 307 and change 

[image]

5) Save the file
6) Refresh the page

ElTh0r0, 2026-09-03:
> hi [@autourdupc](https://github.com/autourdupc) i've a problem with the last file lbo_chart_p.php it' ok on a smartphone screen but on a PC display is to narrow... how could i correct this ? regards

I am **not** using @autourdupc's customized chart view, but I can **confirm the shrinking** shown by @defaliz for the default charts! So I'd assume it is a general RPI-ECU issue (or more likely a bug in an included library --> [chart.js](https://github.com/chartjs/Chart.js/issues?q=is%3Aissue%20shrink)).

I am using v5 same as @defaliz. This may explain, why @autourdupc cannot reproduce the issue who is still on v4.3.

defaliz, 2026-09-03:
It's an update...
Thanks for correction Laurent, i'll try tomorrow..
Good evening

defaliz, 2026-09-04:
bonjour Laurent
i've changed STYLESHEETS files and all is good now
thank a lot
Alain

autourdupc, 2026-09-04:
Hello.

Stylesheets have been modified in new version of RPI-ECU, so using the old ones is definitively the simple way :)