patience4711/RPI-APS-inverters #131: How to get the exact version running ?

autourdupc, 2024-12-06:
Hi !

How can I get the exact version running ?
I mean that I know that the version is "firmware version = RPI-ECU-v4_2", but is it 4.2a, 4.2b or ... ?

I think that the exact version should be displayed in the "info.php" page.

Regards,
Laurent

patience4711, 2024-12-13:
install update 4_3.

autourdupc, 2024-12-15:
Hello @patience4711 

Just updated with V4.3.
Still remains on 4.2 !

[image]

[image]

And just after a reboot; it is not better.
Note also a little confusion on texts 
[image]

patience4711, 2024-12-16:
I checked and all seems oke to me. 
During update a script writes "echo 'RPI-ECU-v4_3' > /var/www/static_data/swVersion.txt"

Can you check that this file is oke? If not, the permissions on that file are probably not r/w.

Or your infopage doesn't read that file. Or you get a cached version of the infopage.

patience4711, 2024-12-16:
Yes i get the point,  the script doesn't write to the file. You didn't aswer my questions.

You can **echo 'RPI-ECU-v4_3' > /var/www/static_data/swVersion.txt**  yourself and see if that works.

autourdupc, 2024-12-17:
It works using the echo cmde...

[image]

patience4711, 2024-12-18:
I just installed image 4_2 on an sd and updated to 4_3. No problems found, the version shown in the infopage = 4_3

autourdupc, 2024-12-18:
Hum... I did not change anything on the 4_2... Just added my personal power script, nothing else !
Very strange

patience4711, 2024-12-18:
Your echo command worked, and that is exactly the command the script runs so we have another inexplicable phenomenon.
In my case the update log showed that the command was issued. I repeated with the file i downloaded on github to be sure. 
So the only possibility is that the browsercache showed the old page.

[image]