patience4711/RPI-APS-inverters #133: SD-CARD speed tests on RPI-ECU - Having a good card make the difference !

autourdupc, 2025-01-06:
Hello all.

I would like to give my experience on sd-card choice and speed.

I was aware that Samsung EVO were a very good choice for reliability and speed, but found that my RPI was not very fast, having some tiles lags on console and on http rendering web pages.
According to this page [https://bret.dk/best-raspberry-pi-micro-sd-cards/#Raspberry-Pi-Zero-2-microSD-Card-Performance](url)
I decided to do some tests.
I use a **Pi zero 2 W** for the RPI-ECU project.

Here are my tests...

### Samsung EVO 16 Plus HCI U3
```
ecuuser@192.168.20.114:~$ dd if=/dev/zero of=./Testing bs=100M count=10 oflag=direct
10+0 records in
10+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 231.504 s, 4.5 MB/s
ecuuser@192.168.20.114:~$ dd if=./Testing of=/dev/zero bs=100M count=10 oflag=dsync
10+0 records in
10+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 239.718 s, 4.4 MB/s
```

[image]

As we can see, R/W speed is **4,5 MB/s**

### Kioxia EXCERIA HCI U1
```
ecuuser@192.168.20.114:~$ dd if=/dev/zero of=./Testing bs=100M count=10 oflag=direct
10+0 records in
10+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 66.1218 s, 15.9 MB/s
ecuuser@192.168.20.114:~$ dd if=./Testing of=/dev/zero bs=100M count=10 oflag=dsync
10+0 records in
10+0 records out
1048576000 bytes (1.0 GB, 1000 MiB) copied, 44.7295 s, 23.4 MB/s
```

[image]

Read speed is **15,9 Mb/s**, Write speed is **23,4 MB/s**

Using Samsung, TOP command always showed a hight **Load average CPU**
And also a hight **waiting’ CPU time** -> wa 
[image]

Using Kioxia, the same TOP command show a low **Load average CPU**
I can note that **waiting’ CPU time** (wa) is now correct
[image]

So my conclusion is that kioxia Exceria sd-card is the best, far from Samsung EVO Plus sd-card.

patience4711, 2025-01-08:
@autourdupc  good job, this gives us more insight in system performance. 

Did you ever consider to install **zswap** to prevent the use of swapfile on the sdcard?
https://linuxblog.io/raspberry-pi-performance-add-zram-kernel-parameters/

I had this running on the old linuxkernel but forgot to install it on the new one. Today i installed with    and added some lines to /etc/sysctl.conf  following the above website. It works without problems.
the steps i followed:
**apt install zram-tools**
**nano /etc/sysctl.conf and add the next 4 lines**
- vm.vfs_cache_pressure=500
- vm.swappiness=100
- vm.dirty_background_ratio=1
- vm.dirty_ratio=50

**sysctl --system**

I think that it can improve the overall system performance as wel as the life of the sdcard.  You might be able to measure the difference.

autourdupc, 2025-01-09:
@patience4711 This is why I told you that something changed in new version because I found that it was too slow...
(https://github.com/patience4711/RPI-APS-inverters/issues/121#issuecomment-2524192353)

Well, according to this link [https://en.wikipedia.org/wiki/Zram](url), it reduce the writes on swap, so I think it a good idea to improve the life of sd-card.

My question is : what Pi are you using ? Waht amount of memory the settings are for ?

Pi zero 2W is only 512 MB of RAM... Are thoses settings suitable for ?

autourdupc, 2025-01-09:
@patience4711 
I installed zram on my Pi zero 2W, working the same way as before.
No change yet visible... I think the enhancement would have been be visibe if I still had the Samung sd-card, but now with the Kioxia sd-card, it works like a charm !

autourdupc, 2025-01-09:
@patience4711 
On guy working on embedded systems using Linux told me that InfluxDB comes very hard to manage as soon there is a write data issue or a corrupt part of the database. In this case, it is impossible to retreive data and most of the time all the database is lost.
He uses now Postgresql and use Timescale extension so he can approch InfluxDB performances, with the ease of sql queries.
Maybe you should have a look at :)

patience4711, 2025-01-10:
Maybe... or maybe not. At the time i considered that the best choice was influx, if i had to change that, it would mean lots of work so i rather not. 
To be honest i discovered a lot of other nice things to do so i I don't want to spend too much time on this project anymore. Only the necessary things.

autourdupc, 2025-01-16:
For those who need more informations to understand the Linux CPU charge details using the "top" command...
[https://unix.stackexchange.com/questions/18918/linux-top-command-what-are-us-sy-ni-id-wa-hi-si-and-st-for-cpu-usage](url)

closed 2025-02-11 by patience4711