patience4711/RPI-APS-inverters #134: frequently checking zb system failed.. why ?

defaliz, 2025-01-12:
Hi,
very often have no connection with zigbee device
nothing has changed in the hardware, i've the last version of rpiecu (never had this problem with version 3.xx)
here are the logs 11:50 it's OK, 12:00 it's fail...
thanks for help

11:50:30 checking zb system running ok
11:50:57 checking zb system running ok
11:51:07 checking zb system running ok
11:52:34 polled inv 1 error 11 AF_DATA_CONFIRM fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
11:55:07 polled inv 0 error 6 inverter not paired
11:55:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
11:59:32 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
11:59:49 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
12:00:13 polled inv 0 error 6 inverter not paired
12:00:14 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
12:00:43 checking zb system failed, an unknown error occurred
12:00:48 checking zb system failed, an unknown error occurred

patience4711, 2025-01-27:
Impossible to say, the zigbee system is equal in all software versions, nothing has changed since version 0.  So it must be something in the air between the inverter and the zigbee antenna. Or the weather is that bad that the inverter doesn't have the power to send messages or resets frequently.

defaliz, 2025-01-27:
Hi Patience
you're right, i've had no problem last summer... but since several months weather is very cloudy with no sun....
regards

closed 2025-02-11 by patience4711