patience4711/RPI-APS-inverters #136: NEW VERSION 4_4

patience4711, 2025-01-27:
This version contains a feature called zram that ensures that swap operations take place in memory instead of on the SD card. Although the software is designed to minimize these write operations, it is possible that zram will provide slightly better speed and sdcard longevity.

Sadly it is too complicated to contain this feature in a software update so there are only 2 ways to obtain it.
- you install the image or 
- follow the instructions in the wiki [here](https://github.com/patience4711/RPI-APS-inverters/wiki/SOFTWARE-UPGRADE/_edit#add-zram)

Please consider that your rpi-ecu will work perfectly without this.  If you have a system with many inverters however, it could be that you experience some improved response speed in the web pages. Otherwise you would probably never know the difference.

kentbrock, 2025-03-01:
I would like to run the image on an RPI2 - it was running for years on the RPI2 until the SD Cards gets defect. I´ve tried two ways:

1) getting to the admin page via ethernet - for this I need valid credentials 
2) getting it running via wifi (stick) - the wifi password rpiecu123 is not accepted

I may need some help to get it running again :-)

patience4711, 2025-03-31:
There are others running the image on an rpi2, it should work. Not sure what the problems are here. What means getting to the admin page via ethernet? And do you mean the admin password to get into the menu? (default 0000).
And should i conclude that you are not able to connect the wifi? Can you log-in via ssh?

kentbrock, 2025-04-05:
The rpi2 was running for years now with the Version <4 in an unchanged HW setup. My SD Card crashed so I´ve tried to get the image 4.4 running. 

SSH login works via LAN
Landing Page of the ECU works via LAN

I´m not able to start the initial configuration via AP because WLAN AP is not connectable due to unkown reasons (I´ve tried diffrent WIFI Sticks, checked ifconfig,...) 
So I tried to skip this and start the configuration directly via LAN.
But its not possible to get to the Menu page because of unknown credentials. 

[image]

kentbrock, 2025-04-11:
I don´t think this is realy hardware related. If it´s possible for me to get into the main menu with the correct credentials I assume everthing would be fine. 

It would be great if you are able to provide me the initial username and password for access the main menu. 

Thanks :-)

patience4711, 2025-05-07:
The credentials are in /etc/apache2/.htpasswd, if you inspect that, it should have one line like:
admin:$apr1$JgA9hb1.$TvbgAO8s/mOfgHJz0awZm.

You can try **admin** and **1234** if this doesnt work, set a new one via the terminal:
**htpasswd -bc /etc/apache2/.htpasswd  user paswd**

closed 2025-05-29 by patience4711