patience4711/RPI-APS-inverters #137: Inverter Error 4

KRLP1, 2025-02-23:
Hallo,

seit heute bekomme ich folgende Fehlermeldung:

15:10:32 checking zb system failed, no answer on serial
15:11:00 checking zb system failed, no answer on serial
15:11:50 checking zb system failed, no answer on serial
setPermissions.sh : checked permissions OK
15:15:10 polled inv 1 error 4 no data received
invdataNull.cgi: wrote a new /ramdisk/invData1 with null values
15:20:08 polled inv 1 error 4 no data received
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
15:25:08 polled inv 1 error 4 no data received
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
15:30:08 polled inv 1 error 4 no data received
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
15:35:08 polled inv 1 error 4 no data received
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1

Wo liegt das Problem? Gestern funktionierte noch alles problemlos.
Ich benutze einen DS3- S als Inverter.

patience4711, 2025-02-24:
Schwer zu sagen, it seems that there is nothing on the serial port, could be the wiring or the zigbee module died. 
So from 15:10 it stopped working en it didn't start the next day? Did you test the serial communication as suggested [here](https://github.com/patience4711/RPI-APS-inverters/wiki/TESTING-AND-DEBUGGING#zigbee-loopback) ?

KRLP1, 2025-02-24:
No i don't test it.
The RPI ECU has actually been in operation for over a year.
It just stopped working for no apparent reason.
Then I tried further.

1. Restart ECU, but the problem still exists
2. Delete and recreate the inverter
3. Try to pair the inverter, but it always fails
4. Power connection and solar panels disconnected from inverter
5. Reconnected everything and tried to pair the inverter, but it failed
6. ECU restarted
7. Tried pairing the inverter, this time it worked and now everything is working again

There were no failures today.
Is there an explanation for what error 4 means?

KRLP1, 2025-03-07:
Today i have a new Problem:

15:42:50 checking zb system failed, an unknown error occurred
15:43:07 checking zb system running ok
setPermissions.sh : checked permissions OK
15:45:08 polled inv 1 1st poll after inverter start
15:45:08 polled inv 1 ts: 3036 signal q: 45.1%
15:50:08 polled inv 1 ts: 300 signal q: 45.1%
15:55:07 polled inv 1 ts: 300 signal q: 45.1%
16:00:07 polled inv 1 ts: 299 signal q: 45.1%
16:05:08 polled inv 1 ts: 301 signal q: 45.1%
16:10:08 polled inv 1 ts: 300 signal q: 45.1%
16:15:10 polled inv 1 ts: 302 signal q: 45.1%
16:20:07 polled inv 1 ts: 297 signal q: 45.1%
16:25:08 polled inv 1 ts: 301 signal q: 45.1%
16:30:07 polled inv 1 ts: 299 signal q: 45.1%
16:35:07 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
16:40:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
16:45:10 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
16:50:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
16:55:08 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1
17:00:07 polled inv 1 error 10 AF_DATA_REQ fail
invdataNull.cgi: set values to null except
for totalen and time in /ramdisk/invData1

I rebooted at 15:42h because the ECU was no longer reading data.
It worked for a while and then the error occurred again.

Output from Zigbee Test:

running script sendZigbee.pl...
argument is 2710AABBCC
sudo /usr/lib/cgi-bin/ecu/testZigbee.cgi 2710AABBCC
Content-type:text/html
running testZigbee.cgi version 2024-sep-2

script testZigbee running with arg 2710AABBCC
serialPort = /dev/serial0 slen = 03

bufferSend + checkSum = FE032710AABBCCE9 got data on serial0
readZigbee
inMessage = FE036710AABBCCA9 command processed
HTTP:1/1 200 OK

Can you Help?

KRLP1, 2025-03-16:
It seems to work now.
I used jumper wires to connect the zigbeemodule, i have soldered the wires now.
Panels look clean and i don't have changed the position of the RPI-ECU.
The Inverter is mounted under the roof, i don't think there will be any corrosion problems. 
Thanks for your help.

closed 2025-03-31 by patience4711