patience4711/RPI-APS-inverters #138: problem with influxd

defaliz, 2025-04-02:
Hi,
i've the last version  RPI-ECU-v4_4 on a rpi zero
but there is a problem with influxd, when i check database with the menu : 
running script healthInflux.pl...
curl -sl -I http://localhost:8086/ping
** health influxdb **
result=
influxdb is currently not (yet) running

yet on the rpi influxd seem to be running, i can see it with the command... "service influxd status"
in fact influx is working only if i launch the command in cli, without service like that : 
influxd

do you know why service doesn't launch correctly influxd ?

regards

defaliz, 2025-04-05:
It's very strange...
when i use "systemctl status influxd"   influxd is running
but the command "netstat -lnp | grep 8086" return nothing and i can't connect to influx
but
when i launch influxd like this : "/usr/bin/influxd -config /etc/influxdb/influxdb.conf"
port 8086 is open and i can connect to influxd...

i really don't understand...

patience4711, 2025-04-18:
An anoying property of influx is that it starts very slow at boot, especially when there is a large database to index [look here](https://github.com/patience4711/RPI-APS-inverters/wiki/DATABASES#important). 

So after some time the healthcheck should say that it is running. But I cannot repoduce this problem so sadly I cannot help you with this.

defaliz, 2025-04-18:
Thanks for your reply ..
Problem is that port 8086 is only open when influxd is started with the command
usr/bin/influxd -config /etc/influxdb/influxdb.conf

And never with systemctl...
Anyway, that works like that...
Regards

closed 2025-05-16 by patience4711